tuxnsk/nodejs_libmodbus #15: gyp: Unable to find targets in build file binding.gyp while trying to load binding.gyp

zachheine, 2015-02-09:
gyp: Unable to find targets in build file binding.gyp while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/Cellar/node/0.10.18/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:424:16)
gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:789:12)
gyp ERR! System Darwin 14.0.0
gyp ERR! command "node" "/usr/local/Cellar/node/0.10.18/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--python=python2"
gyp ERR! cwd /Users/zrh0/Desktop/modbus/node_modules/modbus
gyp ERR! node -v v0.10.18
gyp ERR! node-gyp -v v0.10.9
gyp ERR! not ok 

> modbus@0.0.13 preuninstall /Users/zrh0/Desktop/modbus/node_modules/modbus
> rm -rf libmodbus build

npm ERR! weird error 1
npm ERR! not ok code 0

nzfarmer1, 2015-02-17:
You' on a Mac right?

See:
https://github.com/tuxnsk/nodejs_libmodbus/issues/4

sivatumma, 2015-02-18:
I have windows 7 with Visual Studio 12 tools installed and am facing the same thing. 
python 2.7.6 and node some stable version are in place. Finding this problem since a week now.

nzfarmer1, 2015-02-18:
you would be.  you need to add your definitions to bingind.gyp for windows.  I added custom ones for mac.  not sure what yours would be but when you figure it out please update this issue with your binding.gyp definitions

ProteusCortex, 2015-03-31:
Can you try to build it again?

closed 2016-08-31 by Disminder