tuxnsk/nodejs_libmodbus #18: ERROR Can't create pipe (Too many open files)

jmutanen, 2015-04-27:
When I run modbus example (Creating master device) multiple times in a row on my Raspberry Pi I got finally an error while creating new master context:

ERROR Can't create pipe (Too many open files)

I use createConRtu for con. It looks like some resources are not freed in the close or free functions?

https://www.npmjs.com/package/modbus#creating-master-device-client

asfyra, 2015-04-27:
I have exactly the same problem running the package on a raspberry pi.

cprice26, 2015-04-28:
I am also having this problem after the nodejs server runs for a few minutes.

jmutanen, 2015-04-28:
I quickly looked to the code and it seems that two file descriptions (ctx_mt->mtp_r and ctx_mt->mtp_w) opened in modbus-rtu.c:1097 are never closed.

int mtp[2];
    if (pipe(mtp) == -1) {
        fprintf(stderr, "ERROR Can't create pipe (%s)\n",
			strerror(errno));
		modbus_free(ctx);
        return NULL;
    }
    ctx_mt->mtp_r = mtp[0];
    ctx_mt->mtp_w = mtp[1];

jmutanen, 2015-05-15:
I made a fix for this one, pull request created. Fix available in https://github.com/jmutanen/nodejs_libmodbus.

jmutanen, 2015-05-16:
There was still one issue. I hade to make one small modification.

nodejs_libmodbus/modbus.js::createMasterRtu and api.Destroy function:

mb.close(ctx);
=>
mb.close_mt(ctx);

Should the same modifications be applied in createMasterTcp api.Destroy function?

cprice26, 2015-06-04:
jmutanen, 

I did the following:
1. I replaced my existing mt.patch file with yours. 
2. I replaced my existing make_libmodbus.sh yours. 
3. Then I replaced my modbus.js file with yours. 
4. I then re-built running ./make_libmodbus.sh

I still receive the following errors after it runs for few minutes:
ERROR Can't create pipe (Too many open files)
ERROR createMasterRtu: Too many open files

Did I do something wrong in applying the patches or is there still an issue?

Thank you for your help!

jmutanen, 2015-06-04:
Hi,

Did you build the package with node-gyp?

./make_libmodbus.sh && node-gyp configure && node-gyp build

Cheers,
 -Jussi

cprice26, 2015-06-05:
That was the problem. I did not rebuild with node-gyp. Seems to be working fine now. Thanks!

closed 2016-08-31 by Disminder