tuxnsk/nodejs_libmodbus #22: Node 0.12 compatibility

iMarvinS, 2015-07-29:
Could it be that this awesome module is not comptabile with node v0.12.x ?
I cannot compile it properly.

iMarvinS, 2015-07-29:
Here´s the output from ```node-gyp build```:

make: Entering directory '/root/test/nodejs_libmodbus/build'
  CXX(target) Release/obj.target/modbus_binding/src/main.o
../src/main.cpp:22:32: error: 'Arguments' does not name a type
 Handle<Value> js_new_rtu(const Arguments& args) {
                                ^
../src/main.cpp: In function 'v8::Handle<v8::Value> js_new_rtu(const int&)':
../src/main.cpp:23:33: error: invalid types 'const int[int]' for array subscript
  String::Utf8Value device(args[0]);
                                 ^
../src/main.cpp:24:40: error: invalid types 'const int[int]' for array subscript
  int baud = Local<Integer>::Cast(args[1])->Int32Value();
                                        ^
../src/main.cpp:25:37: error: invalid types 'const int[int]' for array subscript
  String::Utf8Value parity_str(args[2]);
                                     ^
../src/main.cpp:26:44: error: invalid types 'const int[int]' for array subscript
  int data_bit = Local<Integer>::Cast(args[3])->Int32Value();
                                            ^
../src/main.cpp:27:44: error: invalid types 'const int[int]' for array subscript
  int stop_bit = Local<Integer>::Cast(args[4])->Int32Value();
                                            ^
../src/main.cpp:33:31: error: too few arguments to function 'v8::Handle<v8::Primitive> v8::Null(v8::Isolate*)'
  if (ctx == NULL) return Null();
                               ^
In file included from ../src/main.cpp:2:0:
/root/.node-gyp/0.12.4/deps/v8/include/v8.h:306:28: note: declared here
   friend Handle<Primitive> Null(Isolate* isolate);
                            ^
../src/main.cpp:35:9: error: 'Wrap' is not a member of 'v8::External'
  return External::Wrap(ctx);
         ^
../src/main.cpp: At global scope:
../src/main.cpp:40:44: error: 'Arguments' does not name a type
 Handle<Value> js_rtu_get_serial_mode(const Arguments& args) {
                                            ^
../src/main.cpp: In function 'v8::Handle<v8::Value> js_rtu_get_serial_mode(const int&)':
../src/main.cpp:41:42: error: 'Unwrap' is not a member of 'v8::External'
  modbus_t *ctx = static_cast<modbus_t *>(External::Unwrap(args[0]));
                                          ^
../src/main.cpp:41:65: error: invalid types 'const int[int]' for array subscript
  modbus_t *ctx = static_cast<modbus_t *>(External::Unwrap(args[0]));
                                                                 ^
../src/main.cpp:45:25: error: no matching function for call to 'v8::Integer::New(int&)'
  return Integer::New(ret);
                         ^
../src/main.cpp:45:25: note: candidate is:
In file included from ../src/main.cpp:2:0:
/root/.node-gyp/0.12.4/deps/v8/include/v8.h:2012:25: note: static v8::Local<v8::Integer> v8::Integer::New(v8::Isolate*, int32_t)
   static Local<Integer> New(Isolate* isolate, int32_t value);
                         ^
/root/.node-gyp/0.12.4/deps/v8/include/v8.h:2012:25: note:   candidate expects 2 arguments, 1 provided
../src/main.cpp: At global scope:
../src/main.cpp:50:44: error: 'Arguments' does not name a type
 Handle<Value> js_rtu_set_serial_mode(const Arguments& args) {
                                            ^
../src/main.cpp: In function 'v8::Handle<v8::Value> js_rtu_set_serial_mode(const int&)':
../src/main.cpp:51:42: error: 'Unwrap' is not a member of 'v8::External'
  modbus_t *ctx = static_cast<modbus_t *>(External::Unwrap(args[0]));
                                          ^
../src/main.cpp:51:65: error: invalid types 'const int[int]' for array subscript
  modbus_t *ctx = static_cast<modbus_t *>(External::Unwrap(args[0]));
                                                                 ^
../src/main.cpp:52:40: error: invalid types 'const int[int]' for array subscript
  int mode = Local<Integer>::Cast(args[1])->Int32Value();
                                        ^
../src/main.cpp:56:25: error: no matching function for call to 'v8::Integer::New(int&)'
  return Integer::New(ret);
                         ^
[... 1500 lines cut ...]
../src/main.cpp:641:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../src/main.cpp: In function 'v8::Handle<v8::Value> json_to_map(const int&)':
../src/main.cpp:675:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../src/main.cpp: In function 'v8::Handle<v8::Value> tcp_accept_async(const int&)':
../src/main.cpp:722:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
modbus_binding.target.mk:87: recipe for target 'Release/obj.target/modbus_binding/src/main.o' failed
make: *** [Release/obj.target/modbus_binding/src/main.o] Error 1
make: Leaving directory '/root/test/nodejs_libmodbus/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.18.14-2-ARCH
gyp ERR! command "node" "/usr/bin/node-gyp" "build"
gyp ERR! cwd /root/test/nodejs_libmodbus
gyp ERR! node -v v0.12.4
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok

scsirdx, 2015-07-30:
Same error.

jeremypoulter, 2015-08-13:
Yes, looks like this module only supports node.js < 0.11. Unfortunately looks like there are major changes to the node.js (V8) internal APIs between 0.10 and 0.12.

jkeiter, 2015-08-19:
I have the following using armhf and compiling on a RPI 1 with latest updated raspbian for the first function in main.cpp
Handle<Value> js_new_rtu(const Arguments& args) { ..

gyp info it worked if it ends with ok
gyp info using node-gyp@2.0.2
gyp info using node@0.12.6 | linux | arm
gyp info spawn make
gyp info make: Entering directory '/home/pi/nodeLibmodbus-master/build'
spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/modbus_binding/src/main.o
../src/main.cpp:24:32: error: ‘Arguments’ does not name a type
../src/main.cpp:24:43: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
../src/main.cpp: In function ‘v8::Handle<v8::Value> js_new_rtu(const int&)’:
../src/main.cpp:25:33: error: invalid types ‘const int[int]’ for array subscript
../src/main.cpp:26:40: error: invalid types ‘const int[int]’ for array subscript
../src/main.cpp:27:37: error: invalid types ‘const int[int]’ for array subscript
../src/main.cpp:28:44: error: invalid types ‘const int[int]’ for array subscript
../src/main.cpp:29:44: error: invalid types ‘const int[int]’ for array subscript
../src/main.cpp:35:31: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Null(v8::Isolate*)’
/home/pi/.node-gyp/0.12.6/deps/v8/include/v8.h:306:28: note: declared here
../src/main.cpp:37:9: error: ‘Wrap’ is not a member of ‘v8::External’
../src/main.cpp: At global scope:
../src/main.cpp:42:44: error: ‘Arguments’ does not name a type // and so on ..

I dont have a lot of experience in c++ debug but the Arguments type error seems to be the first issue..
maybe a compiler switch? Perhaps the compiler update is the issue and not V8.
Thanks

jeremypoulter, 2015-08-19:
Sorry, not a compiler issue this is a well known issue with Node 0.12.

Probably the best way forward is to port to use

jeremypoulter, 2015-08-19:
Oops wrong button...

https://github.com/nodejs/nan

I am going to try and look at this so if anyone wants to help or has already started let me know

jkeiter, 2015-08-22:
Ok it is not a compiler issue.. Seems looking at the node JS 0.12 addon page that this driver needs a substantial rewrite.. 
for instance I can't find an example for returning the CTX structure if based on the existing code:
Handle<Value> js_new_rtu(const FunctionCallbackInfo<Value>& args) { 
	Isolate* isolate = Isolate::GetCurrent();
	HandleScope scope(isolate);
        //
I find examples showing for instance :
void  js_new_rtu(const FunctionCallbackInfo<Value>& args) {
        Isolate* isolate = Isolate::GetCurrent();
	HandleScope scope(isolate);
       //
so the CTX structure has to be wrapped in the isolate object..??
or the node_object_wrap class object.. i.e.
    MyObject* obj = new MyObject(value);
    obj->Wrap(args.This());
    args.GetReturnValue().Set(args.This());

So I'm thinking the biggest hurdle for me right now is to wrap the ctx object and unwrap it in the correct way using
void ####(const FunctionCallbackInfo<Value>& args)

I was looking at https://github.com/nodejs/node-addon-examples  
8_passing_wrapped/node_0.12

PaulCharlton, 2016-01-23:
see https://github.com/tuxnsk/nodejs_libmodbus/commit/ecb970eb36dd6017269fd6432efb83e882ad83d6

it is a port to the new node native API

jkeiter, 2016-02-01:
> see ecb970e
>
> it is a port to the new node native API

OK I am happy to see this contribution; On an PI 2 I had no issues when
compiling other than the same issues I had before .. ( modbus version and
response timeout which I commented out ) however on execution my node test
js program I get:
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without
a HandleScope
Aborted

FYI
node -v
v5.4.1
npm -v
3.5.3
node-gyp -v
v3.2.1

closed 2016-08-31 by Disminder