tuxnsk/nodejs_libmodbus #25: install modbus not working

brabant-v, 2016-02-04:
I can't seem to get modbus installed 
tried various things from the most obvious google searches but none seem to help.
[image]

num-lock, 2016-02-04:
Taken from the `README.md`:

> Now working only in *nix system.

Windows is out of the scope. (Which is what you are apparently using)

brabant-v, 2016-02-04:
I was doubting if this was the issue but wasn't shure.  Thanks for the confirmation

num-lock, 2016-02-05:
To be honest, your actual problem appears to be that you are most likely missing the build tools required for node-gyp to work. But even if you had them you'd be out of luck.

brabant-v, 2016-02-05:
I've installed a linux machine last night but still encountering the same errors.
there's a problem with the npm installer.

brabant-v, 2016-02-05:
[image]

didn't read it the eroors carefully enough "this is most likely a problem with the modbus package"

num-lock, 2016-02-05:
That's the standard message when the install script is failing, which is trying to call node-gyp at some point. If your would really read your error messages carefully, you'll notice that you are even missing the `git` command.

brabant-v, 2016-02-05:
I've only been using this since monday hence the beginners mistakes.
You have any idea what i can do to make this work (starting with installing git)

num-lock, 2016-02-05:
You need `git` and everything necessary for [node-gyp](https://github.com/nodejs/node-gyp) obviously. This package is just a wrapper for another library (libmodbus) after all.

tuxnsk, 2016-02-06:
install git previously

closed 2016-08-31 by Disminder