tuxnsk/nodejs_libmodbus #30: Random data when reading with native connection

acrive82, 2016-06-11:
Hi 

I'm trying to connect to modbus with native connection. Many times the data returned is totally randomize. 

If I try write a value and after to read it, have a random value.

Disminder, 2016-06-12:
Modbus TCP, yes?

It's probably error in libmodbus. I'm gonna to rewrite bindings to latest libmodbus version soon, just don't have enough time now.

acrive82, 2016-06-13:
Yes! With TCP.

Disminder, 2016-07-04:
Rewrite to latest libmodbus version didn't help =_=
Gonna make some experiments - I just didn't understand, why M libmodbus - S libmodbus works and M libmodbus - S real device - not.

Disminder, 2016-07-05:
Not libmodbus error - you need to use `native.set_slave(ctx, id)` function to set which ID you requesting on the slave device. It helped me. Probably I'll add 0 as a default ID.

closed 2016-07-05 by Disminder