tuxnsk/nodejs_libmodbus #33: Slave ID

reejinbouk, 2016-08-12:
Hello, 
how to create modbus slave with predefined Slave ID?

can't find it at examples or docs

thank you

Disminder, 2016-08-12:
I saw text of this issue in mail, I'll extend examples ;)

reejinbouk, 2016-08-13:
Hello
I've tried to create slave

in this case it working:
`'use strict';

var log = console.log;
var mb = require('modbus').create();

mb.onError(function (msg) {
  log('ERROR', msg);
});

// create device memory map
var data = mb.createData({ countReg: 5, countBit: 2, countInputReg: 4, countInputBit: 2, });
data.setReg(2, 321);
data.setBit(1, true);
data.setInputReg(1, 123);
data.setInputBit(0, true);
data.dumpData(); // show memory map

// create slave device
var ctx = mb.createSlave({

  // connection type and params
  con: mb.createConTcp('192.168.10.17', 502),

  // data map
  data: data,

  // callback functions
  onQuery: function () {
    log('onQuery');
  },
  onDestroy: function () {
    log('onDestroy');
  }
});`

in this case it's not working:
`'use strict';

var log = console.log;
var mb = require('modbus').create();

mb.onError(function (msg) {
  log('ERROR', msg);
});

// create device memory map
var data = mb.createData({ countReg: 5, countBit: 2, countInputReg: 4, countInputBit: 2, });
data.setReg(2, 321);
data.setBit(1, true);
data.setInputReg(1, 123);
data.setInputBit(0, true);
data.dumpData(); // show memory map

// create slave device
var ctx = mb.createSlave({

  // connection type and params
  con: mb.createConTcp('localhost', 502),

  // data map
  data: data,

  // callback functions
  onQuery: function () {
    log('onQuery');
  },
  onDestroy: function () {
    log('onDestroy');
  }
});`

i've tried also change localhost to "127.0.0.1" - it stil not working

Disminder, 2016-08-16:
Hi! I'll test it as soon as possible, but I have some code like yours, and it works. Probably, you need to switch port to `1502` or run it with `sudo`, because, as I know, ports < 1000 not allowed for non-sudo users.

reejinbouk, 2016-08-19:
Hi, i've tested with sudo and 1502 port.
still the same - not working (

reejinbouk, 2016-08-19:
Please, write, how to catch commands, receiving from modbus?
I'm successfully write to Bit and Regitsters, can see changes.
But how i can catch this commands?

reejinbouk, 2016-08-20:
Fix it - changed to '0.0.0.0' and it open port for all connection (not only from localhost)

closed 2016-08-20 by reejinbouk

Disminder, 2016-08-31:
Sorry, gmail thought that the letters from the github is spam =/
Something really strange - probably it is error in your system - I can't reproduce this bug. Not for localhost <-> localhost, nor for localhost <-> different PC