tuxnsk/nodejs_libmodbus #34: Bindings for rtu_get_rts/rtu_set_rts commented out

jcwren, 2016-08-14:
I'm using a RS-485 adapter that requires RTS to be managed for transmitting. libmodbus supports this with the modbus_rtu_set_rts() function, but the bindings are commented out in src/main.cpp.

Are these function commented out because of a problem? If you need a way to test them, I have a small piece of C code (using libmodbus) that retrieves a few registers from a slave, and prints them out. I can test a similar piece of nodejs code for verification, if you like.

I'm also hoping that perhaps you can update nodejs_libmodbus to use the 3.1.4 version (latest) of libmodbus.

markdegrootnl, 2016-08-16:
See #35

Disminder, 2016-08-16:
If you can - test this pull request please: https://github.com/tuxnsk/nodejs_libmodbus/pull/35
I don't have a devices, where I can test this.
Btw, libmodbus in this project is 3.1.4, if you install it from github, not from npm.

Disminder, 2016-10-24:
Done

closed 2016-10-24 by Disminder