tuxnsk/nodejs_libmodbus #39: Does not work on Mac OS X 10.11.6

robogeek, 2016-11-26:
I've read through #4 BTW ... I have a similar issue, but with 10.11.6.  With the "Create slave device" sample I get this output:

```
$ node index.js 
dyld: lazy symbol binding failed: Symbol not found: _modbus_mapping_new
  Referenced from: /Users/david/nodejs/modbus-server/node_modules/modbus/build/Release/modbus_binding.node
  Expected in: dynamic lookup

dyld: Symbol not found: _modbus_mapping_new
  Referenced from: /Users/david/nodejs/modbus-server/node_modules/modbus/build/Release/modbus_binding.node
  Expected in: dynamic lookup

Trace/BPT trap: 5
```

During "npm install" I don't see anything suggesting possible problems - there are a couple warnings at the end.

```
$ npm install modbus --save
npm WARN package.json modbus-server@1.0.0 No description
npm WARN package.json modbus-server@1.0.0 No repository field.
npm WARN package.json modbus-server@1.0.0 No README data
\
> modbus@0.0.14 install /Users/david/nodejs/modbus-server/node_modules/modbus
> ./make_libmodbus.sh && node-gyp configure && node-gyp build

Cloning into 'libmodbus'...
remote: Counting objects: 4402, done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 4402 (delta 3), reused 0 (delta 0), pack-reused 4393
Receiving objects: 100% (4402/4402), 1.52 MiB | 1.33 MiB/s, done.
Resolving deltas: 100% (2955/2955), done.
Checking connectivity... done.
HEAD is now at 79a7c1e Fix missing argument in synopsis section of modbus_rtu_set_serial_mode
patching file src/modbus.c
patching file src/modbus.h
patching file src/modbus-private.h
patching file src/modbus-rtu.c
patching file src/modbus-rtu-private.h
patching file src/modbus-tcp.c
patching file src/modbus-tcp-private.h
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /opt/local/bin/aclocal --force 
autoreconf: configure.ac: tracing
autoreconf: running: /opt/local/bin/glibtoolize --copy --force
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
glibtoolize: and rerunning glibtoolize and aclocal.
glibtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: running: /opt/local/bin/autoconf --force
autoreconf: running: /opt/local/bin/autoheader --force
autoreconf: running: /opt/local/bin/automake --add-missing --copy --force-missing
configure.ac:66: installing './compile'
configure.ac:49: installing './config.guess'
configure.ac:49: installing './config.sub'
configure.ac:28: installing './install-sh'
configure.ac:28: installing './missing'
src/Makefile.am: installing './depcomp'
tests/Makefile.am:40: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /opt/local/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /opt/local/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-apple-darwin15.6.0
[... 189 lines cut ...]
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-am
  CC       bandwidth-server-one.o
  CCLD     bandwidth-server-one
  CC       bandwidth-server-many-up.o
  CCLD     bandwidth-server-many-up
  CC       bandwidth-client.o
  CCLD     bandwidth-client
  CC       random-test-server.o
  CCLD     random-test-server
  CC       random-test-client.o
  CCLD     random-test-client
  CC       unit-test-server.o
  CCLD     unit-test-server
  CC       unit-test-client.o
  CCLD     unit-test-client
  CC       version.o
  CCLD     version
make[2]: Nothing to be done for `all-am'.
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
  CXX(target) Release/obj.target/modbus_binding/src/main.o
  SOLINK_MODULE(target) Release/modbus_binding.node
clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9
```

Disminder, 2016-11-26:
Try to install it from git, not from npm, because mom now contain old version.
Then try ’brew install libmosbus’

robogeek, 2016-12-05:
I finally got back to this.  I did: `git clone git@github.com:tuxnsk/nodejs_libmodbus.git` and then in the directory with my test code did `npm install ../nodejs_libmodbus/`

The install seemed to have run correctly ... but

```
$ node index.js 
before createData
Segmentation fault: 11
```

The code is:

```
ar mb = require('modbus').create();

// create device memory map
console.log('before createData');
var data = mb.createData({ countReg: 5, countBit: 2 });
console.log('after createData');
data.dumpData();
data.setReg(2, 321);
data.setBit(1, true);
data.dumpData(); // show memory map
...
```

Therefore the crash happens in `mb.createData` and is different from the crash reported above.